micropython/stm32lib #2: Linker error with PYBBOARD using "...+H7-1.2.0..."

rolandvs, 2018-02-25:
When using the `F4-1.16.0+F7-1.7.0+H7-1.2.0+L4-1.8.1` version, the latest mpy doesn't compile anymore without errors:

```
...
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
build-PYBV11/dma.o: In function `dma_init':
dma.c:(.text.dma_init+0xf0): undefined reference to `DMA_CalcBaseAndBitshift'
make: *** [build-PYBV11/firmware.elf] Error 1
```

dpgeorge, 2018-02-26:
Yes, you can't use the vendor branch directly with MicroPython.  You need to use a "working" branch which has some patches applied to the vendor branch to make it work with MicroPython.  There currently is no working branch for the new H7 HAL.  This will come after the H7 is properly working.

rlnktt, 2018-10-06:
@dpgeorge I'm also facing a similar issue while trying to build. It would be of a great help, if you could give some clarity on how to get this done(may be a link or a tutorial) as I am completely new to this concept of vendor branches!!

rlnktt, 2018-10-06:
@dpgeorge Got this fixed by changing altering the git config file!!

Thanks !!

dpgeorge, 2019-04-23:
There is now a working branch for the H7.

closed 2019-04-23 by dpgeorge