micropython/stm32lib #3: [stm32h743] Errata version Y

rolandvs, 2018-10-24:
Dear Damien,

I noticed a bit reversal in the version Y Errata sheet of the STM32H743.
How to cope with such "features" as they vanish with the next silicon revision?

```
#define RCC_BDCR_LSEDRV_Pos                    (3U)
#define RCC_BDCR_LSEDRV_Msk                    (0x3U << RCC_BDCR_LSEDRV_Pos)   /*!< 0x00000018 */
#define RCC_BDCR_LSEDRV                        RCC_BDCR_LSEDRV_Msk
#define RCC_BDCR_LSEDRV_0                      (0x2U << RCC_BDCR_LSEDRV_Pos)   /*!< ERRATE BIT REVERSED 0x00000010 */
#define RCC_BDCR_LSEDRV_1                      (0x1U << RCC_BDCR_LSEDRV_Pos)   /*!< ERRATE BIT REVERSED 0x00000008 */

// SEE ERRATE SHEET REV Y CHIPS ORIGINAL
//#define RCC_BDCR_LSEDRV_0                      (0x1U << RCC_BDCR_LSEDRV_Pos)   /*!< 0x00000008 */
//#define RCC_BDCR_LSEDRV_1                      (0x2U << RCC_BDCR_LSEDRV_Pos)   /*!< 0x00000010 */

#define RCC_BDCR_LSECSSON_Pos                  (5U)
#define RCC_BDCR_LSECSSON_Msk                  (0x1U << RCC_BDCR_LSECSSON_Pos) /*!< 0x00000020 */
```

rolandvs, 2020-01-17:
Is it possible to update the HAL to the latest version 1.6. There seems to be some improvements that also have to do with the die change revision V. I have some problems with the Flash with the Rev V that did not occur in the Rev Y. Especially the Flash gets corrupted after programming the image.

dpgeorge, 2020-01-20:
Yes it would be good to update it.  Hopefully the update includes support for 480MHz SYSCLK as well.

dpgeorge, 2020-07-19:
The H7 HAL is now updated to v1.6.0, currently on branch `work-F0-1.9.0+F4-1.16.0+F7-1.7.0+H7-1.6.0+L0-1.11.2+L4-1.8.1+WB-1.6.0`

closed 2020-07-19 by dpgeorge