micropython/stm32lib #10: [STM32WB55] update version

dmazzella, 2020-07-10:
Hi @dpgeorge 
ST has released update for WB55 hal at this url: [STM32CubeWB](https://github.com/STMicroelectronics/STM32CubeWB).

The reason for the request is related to the firmware of the radio part.
With version HAL 1.1 currently supported by micropython it is possible to install version 1.3 at most.

Is the update planned?

Grazie,
D.

dpgeorge, 2020-07-10:
> With version HAL 1.1 currently supported by micropython it is possible to install version 1.3 at most.

I'm not sure, but it could be that it just works, that you can just use v1.3 radio firmware without any other changes.  MicroPython does not use the HAL to interface to the radio CPU, it uses its own implementation (in `ports/stm32/rfcore.c`), so updating the HAL probably will not do anything.  If there are any changes needed for v1.3 radio firmware then they'll need to be made to `ports/stm32/rfcore.c`.

dpgeorge, 2020-07-19:
The WB HAL is now updated to v1.6.0 from the git repo linked above (v1.7.0 had no changes that would affect the repo here), see ff5b5358e7ffd6a4a849740b673b2612d0246af9.  Working branch is currently `work-F0-1.9.0+F4-1.16.0+F7-1.7.0+H7-1.6.0+L0-1.11.2+L4-1.8.1+WB-1.6.0`

closed 2020-07-19 by dpgeorge

dmazzella, 2020-07-20:
Thanks for update 😉