micropython/stm32lib #23: Reserved bits written in RCC_AHB/APB_FORCE_RESET()

iabdalkader, 2022-12-03:
https://github.com/micropython/stm32lib/blob/0cbbf16b4268904afdcbdf06e73c0c6e64a8f9ae/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h#L4595

Just a heads up, all of the `RCC_<AHB/APB>_FORCE_RESET` macros in this HAL version (1.6?) write reserved bits that shouldn't be written (same for all AHBx APBx reset macros), this was fixed in more recent HALs, for example updated AHB3 reset macro now checks for device revision and write specific bits:

```c
#if (STM32H7_DEV_ID == 0x450UL)
#define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = 0x00015031U)  /* Resets MDMA, DMA2D, JPEG, FMC, QSPI and SDMMC1 */
#elif  (STM32H7_DEV_ID == 0x480UL)
#define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = 0x01E95031U)  /* Resets MDMA, DMA2D, JPEG, FMC, OSPI1, SDMMC1, OSPI2, IOMNGR, OTFD1, OTFD2 and GFXMMU */
#else
#define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = 0x00E95011U)  /* Resets MDMA, DMA2D, FMC, OSPI1, SDMMC1, OSPI2, IOMNGR, OTFD1, OTFD2 */
#endif /* STM32H7_DEV_ID == 0x450UL */
```

Setting all bits, causes at least one issue that I'm aware of when calling `HAL_DeInit` with AHB3 in particular.

dpgeorge, 2022-12-20:
I think the best thing to do is simply update the H7 HAL.

dpgeorge, 2022-12-20:
> I think the best thing to do is simply update the H7 HAL.

That's now done, although there's no branch yet with this new version.

iabdalkader, 2022-12-21:
That should fix it, thanks! Newer HALs also add/improve support for dual core H747, I will be sending some patches later to Portenta's `board_init.c` to add support for proper standby mode with CM7/CM4.

closed 2022-12-21 by iabdalkader